pku-liang/hlcd-spmm-project #5: RhsBuffer 接收新的数据？

DFruoyue, 2024-12-18:
[image]

截图如上所示。

>   最开始，rhs 的 buffer 是空的，阵列首先读入 rhs。当 rhs 读入完成，rhs buffer 非空的时候，允许输入 lhs。输入 lhs 后立刻开始计算，并在一段时间后放到 output buffer 里面。

这是否意味着一旦 **RhsBuffer** 一次读入完成后, 不会再接收新的数据? 如果不是, 如何判断 **RhsBuffer** 是需要使用原有的数据, 还是接收新的数据呢? 是根据 `lhs_ws` 判断吗？

但是 testbench 中的 `lhs_ws` 被放在 lhs数据 中, 只有在 **RhsBuffer** 准备好之后才能获取. 但是, RhsBuffer要准备好, 需要知道是否为`sta` , 锁死了？

KEKE046, 2024-12-18:
默认情况下，也就是没有任何 stationary 的情况下，计算完 rhs 就被删除了，也就是 rhs buffer 将会 ready。下一次输入的 rhs 将会覆盖原来的矩阵，成为新的右矩阵。

在 weight stationary 的情况下，lhs_ws 决定了 rhs buffer 是否被删除。但当 lhs_ws 为高的时候，rhs buffer 不会清除。也就是说，下一次稀疏矩阵乘法将使用和这一次一样的 rhs。如果 lhs_ws 还是为高，那么 rhs 会保留到下下一次。

KEKE046, 2024-12-18:
af93b32 中，把 PE Array 的时序图重新画了一下。

DFruoyue, 2024-12-19:
也就是说 **这一次** 输入的 lhs_ws 会决定 **下一次** 计算是使用当前的 Rhs_buffer 内的数据还是导入新的数据计算是吗

DFruoyue, 2024-12-19:
> [af93b32](https://github.com/pku-liang/hlcd-spmm-project/commit/af93b329f76e20ed61dff6678e1f8b24aab243ac) 中，重新画了 PE Array 的时序图，这可能能清楚一点。

更新的时序图中没有lhs_ws信号，能否展示一个lhs_ws为 0 和 为 1 两种情况都存在的时序图？

KEKE046, 2024-12-19:
191e49b 里放了一个 WS 的时序图

[image]

KEKE046, 2024-12-20:
上面的波形图有 bug，#8 修复了上面波形图的 bug。

KEKE046, 2024-12-21:
如果没有问题，我把 issue 关闭了

closed 2024-12-21 by KEKE046