pku-liang/hlcd-spmm-project #13: output stationary中的输出控制问题

Gracfusie, 2024-12-20:
output stationary中，计算加到上一次的输出矩阵中，但是上一次的运算没有得到任何关于os的信息，理应正常ready并输出。但是test程序认为两次output应当合起来输出，这是否合理？
希望给出output stationary的波形示意图。

Gracfusie, 2024-12-20:
建议：如果将os同ws对齐，改为下次计算把输出加入到这次的输出矩阵，可能会让逻辑直接一些。

KEKE046, 2024-12-21:
project 中没有将 os 和 ws 对齐，一个考虑是降低硬件设计的难度，状态机会稍微简单一点（可能取决于实现）。如果要更改，也只能是明年改，今年应该不会有变化。
[image]

Gracfusie, 2024-12-21:
在os_onepass中，有2个rhs和2个lhs，而out只收集了一次，这需要阻断第一个计算完成后的out_ready，让硬件在第二次计算完成中才给出out_ready。上图中好像没有体现出这一点？

DFruoyue, 2024-12-21:
这个lhs-os的实现应该和outbuffer内部实现无关吧。为什么要考虑OutdbBuffer里面“第二个”之类的呢？

        简单起见 以没有实现db的buffer为例，在这一轮计算完成后 在没有输出结果之前是不能接受下一轮lhs数据的 包括lhs-os 这个时候怎么知道这一轮计算结果是否输出？

       难道第二轮计算中的lhs-os信息能反过来决定上一轮的计算结果是否输出？ 个人认为这个逻辑违反因果律。

KEKE046, 2024-12-22:
这里的 “第二个” 指的是 dbbuf 两个 buffer 中的 第二个。我重新画一个图，应该解释起来更方便。

[image]

KEKE046, 2024-12-22:
在这个的基础上，如果 os 占用了 outbuf 的队尾，那 outbuf 的队尾就不能被输出，因为它已经被占用了。如果 out buf 中只有一个矩阵，那 out_ready 就应该为 0。但如果 out buf 有两个矩阵，那第一个矩阵还是可以输出的，out_ready 为 1。

KEKE046, 2024-12-22:
关于 out_ready 的问题，这取决于 out_start 先输入，还是 lhs_start + os 先输入。如果是后者先输入，且 out buf 中只有一个矩阵，那这个矩阵将被 output stationary 占用。out_ready 将变成 0，等到 os 算完后再变成 1。

KEKE046, 2024-12-22:
最后是 out_start 和 lhs_start + os 竞争抢 out buf 的情况，优先给 lhs_start + os 使用。

DFruoyue, 2024-12-22:
那么如果outbuf没有实现db，是否就无法实现lhs-os功能了

Gracfusie, 2024-12-22:
那意思就是test里面，保证不会出现“os的send_lhs之前有一个receive_out”的情况，对吗

KEKE046, 2024-12-22:
可以这么认为

DFruoyue, 2024-12-22:
但是当输出栈中两个Buffer都满了的时候（也就是已经有两个计算结果等待输出），这个时候不能允许产生新的计算结果，需要将lhs_ready拉低，这个时候 lhs_start + os 会被阻塞，也就无法形成out_start  和 lhs_start + os 竞争

DFruoyue, 2024-12-22:
难道说，已经存在两个计算结果等待输出的时候就无法使用 output-stationary 功能吗？

DFruoyue, 2024-12-22:
> 但是当输出栈中两个Buffer都满了的时候（也就是已经有两个计算结果等待输出），这个时候不能允许产生新的计算结果，需要将lhs_ready拉低，这个时候 lhs_start + os 会被阻塞，也就无法形成out_start 和 lhs_start + os 竞争

因为如果此时还将 lhs_ready 置高，万一输入的 lhs_os 为 0 那么产生的新的计算结果将无法放入Outbuffer 中。

DFruoyue, 2024-12-22:
[image]
比如说单个OutBuffer，其流程图如上所示，在第一轮计算结果尚未输出前，lhs_ready是会被拉低的，此时系统无法接受到下一轮lhs的信息，这种情况下卡死了

KEKE046, 2024-12-22:
lhs ready 是有多个的，可能 ns 和 ws 不是 ready，但 os 是 ready 的。在单个 OutBuffer 的情况下，lhs_ready_os 可以为高，但 lhs_ready_ns 为低。

https://github.com/pku-liang/hlcd-spmm-project/blob/7625fe43b5bb92b54657931ab1b6ff494af96b66/SpMM.sv#L82-L85

DFruoyue, 2024-12-22:
原来如此！

KingRayCao, 2024-12-22:
也就是说，在out_buffer中有值没输出的情况下，就不允许下一次是ns是吗

KEKE046, 2024-12-22:
是的，可以这样认为

closed 2024-12-23 by Gracfusie