pku-liang/hlcd-spmm-project #8: 有关rhs_start使能时机的疑问

KingRayCao, 2024-12-20:
请问测试中`out_start`能够保证一检测到`out_ready`就启动吗？
如若不然，如此处的时序图：

> [1b23e73](https://github.com/pku-liang/hlcd-spmm-project/commit/1b23e738b9cbafbe96b48bfb5dea0b2e1be188a2) 里放了一个 WS 的时序图
> 
> [image]

假设`out_start`一直不使能，则模块需要一直存储着第一次`out`结果。但是前期输入了三组`lhs`数据（此时无法预知后续`out_start`情况），即使是`double buffer`也无法存储三组`out`输出。
而如果能够保证这一点，则在时序上后输入的`lhs`所产生的结果一定严格在前输入的`lhs`之后，应该不需要`double buffer`？

KEKE046, 2024-12-20:
out_start 可能一直不使能，这样会阻塞，导致 lhs 和 rhs 都不能 ready

KingRayCao, 2024-12-20:
那这样的话上面的样例不是有问题吗？在`lhs_os`未使能时，不能一次性读三个lhs进来

KEKE046, 2024-12-20:
确实，这幅图可能有问题，你可以认为中间有 output stationary，我可以改一下，把 out 往前拖一点

KEKE046, 2024-12-20:
ab21921 中修改了时序图

[image]

KEKE046, 2024-12-21:
如果没有问题，我把 issue 关闭了

closed 2024-12-21 by KEKE046