pku-liang/hlcd-spmm-project #15: SpMM测试中矩阵的维度

angerybob, 2024-12-21:
[image]
[image]

请问为什么make N=16 SpMM后测试样例显示的矩阵维度是19*19的，并且仿真波形中lhs_data前三个数一直是16、17、18？

KEKE046, 2024-12-22:
你是否修改了 SpMM 中的 assign，testbench 是通过这条语句来识别 N 的。

https://github.com/pku-liang/hlcd-spmm-project/blob/7625fe43b5bb92b54657931ab1b6ff494af96b66/SpMM.sv#L103

另外，也可能是没有 `make clean`，可以 clean 之后重新试试。

angerybob, 2024-12-22:
> 你是否修改了 SpMM 中的 assign，testbench 是通过这条语句来识别 N 的。
> 
> https://github.com/pku-liang/hlcd-spmm-project/blob/7625fe43b5bb92b54657931ab1b6ff494af96b66/SpMM.sv#L103
> 
> 另外，也可能是没有 `make clean`，可以 clean 之后重新试试。

确实，我把num_el和delay弄混了，修改后正常了，感谢！

closed 2024-12-22 by angerybob