pku-liang/hlcd-spmm-project #18: 有关最终输出异步/同步的疑问

KingRayCao, 2024-12-22:
请问最后的输出，是要求`out_start`一被激活就立刻输出数据，还是等待`out_start`信号被时钟沿采集到再开始输出？两者会导致一个时钟周期的延迟。
按照tb的代码，似乎是前者，但是个人认为后者更加合理。
https://github.com/pku-liang/hlcd-spmm-project/blob/7625fe43b5bb92b54657931ab1b6ff494af96b66/SpMM.tb.cpp#L246-L260

KEKE046, 2024-12-23:
我觉得是现在的更合理。这里是模拟了 ready valid 接口，out_ready 可以看作是 "valid"，out_start 看作是 "ready"。在输出准备好的时候，输出的部分就应该先 "valid" (out_ready=1)，out_data 也应该填上第一行的信号，发送给后面的 module。后面的 module 收到信号后，可能会返回一个 "ready"，也就是 out_start=1。

KingRayCao, 2024-12-23:
但是如果start和时钟沿对齐的话，也会有很奇怪的行为，因为输出计数器还是得在时钟沿进行更新

KEKE046, 2024-12-23:
我觉得没有问题啊，不管哪种，计数器都是在时钟沿更新的。

```systemverilog
reg out_started;
assign out_en = out_start || out_started;
assign out_ctr_next = out_en ? out_ctr + 1 : out_ctr;
always_ff @(posedge clock)
  if(reset) out_started <= 0;
  else out_started <= out_start || (out_ctr != (`N/4-1));
```

KingRayCao, 2024-12-23:
但是我有大量的逻辑是依靠同步的方式进行更新，如果需要考虑异步的读出的话状态转移会非常难写。
能否允许同步输出呢？只需要将tb中的step()移到循环开头即可。

KingRayCao, 2024-12-23:
> 我觉得没有问题啊，不管哪种，计数器都是在时钟沿更新的。
> 
> ```systemverilog
> reg out_started;
> assign out_en = out_start || out_started;
> assign out_ctr_next = out_en ? out_ctr + 1 : out_ctr;
> always_ff @(posedge clock)
>   if(reset) out_started <= 0;
>   else out_started <= out_start || (out_ctr != (`N/4-1));
> ```

这个逻辑还是有问题的，比如N=4的时候，out_started本来应该永远为0（因为一共只有4行，在第一个时钟沿前就已经输出完毕，cnt不会超过1）但由于时钟沿时out_start为高，out_started会被拉成高，假设cnt不会溢出回到0的话甚至会死循环

KEKE046, 2024-12-24:
可以特判一下吧，N=4 的时候，ctr 会成为一个宽度为 0 的寄存器，这个时候退化成什么都不作。

closed 2024-12-26 by KEKE046