pku-liang/hlcd-spmm-project #20: 关于os的逻辑—os one pass样例

zZzZ9zZ9, 2024-12-24:
如图所示的测试波形图
[image]

在测试中会发现，只要在第一个矩阵计算结束后，在开始输入第二个lhs，即保证第一个out_ready信号在第二个lhs_ready_ns前面，就可以保证此时的out_start不会开始。

但这种逻辑不符合之前的波形图要求，波形图中是要求lhs的输入是能够被overlap掉的。
[image]

所以可能需要重新修改testbench代码，如果后面有os的需求，那么应该跳过接下来的第一个out_ready信号，并不输出（不给out_start）。或者就算输出了也没事，知道目前接收的输出是部分和就行。

zZzZ9zZ9, 2024-12-24:
out_ready的信号应该是完全依赖于lhs的输入的，lhs输入了几个就有几个out_ready信号。应该在testbench里面自主选择哪些out_ready之后要out_start，out_ready的个数是和LHS个数完全一致的

KEKE046, 2024-12-24:
ok，我明白你的意思，三个接口完全异步的 tb。目前的 tb 主要是为了方便大家 debug，都是模块化的简单情况。更复杂的情况将在 27 日评测时同时放出来。

zZzZ9zZ9, 2024-12-24:
嗯好的，但是我的意思是目前的testbench是不是就是错的？能不能修改一下（）。应该修改了让testbench自主选择哪个output是需要的。
比如在这个样例中，如果只要第二个os的结果，那么第一个start就该跳过。

zZzZ9zZ9, 2024-12-24:
目前的testbench是这样的。
[image]
只要保证了第二个lhs在第一次输出完毕后再输入就没有问题。样例中给的所有case都能通过。
[image]
但我觉得这种可能不太对，相当于输入lhs的时间完全没有被overlap掉。那batch_size特别大的时候，这个延时长太多了。

zZzZ9zZ9, 2024-12-24:
[image]
之前的第一张波形图忘记放注释了，实现的是图中这样的逻辑。里面两个lhs_start是紧挨着的，第二个的lhs输入已经被第一个的计算过程和输出过程overlap掉了

KEKE046, 2024-12-26:
我理解了，中期评测会重新写一下 tb

zZzZ9zZ9, 2024-12-26:
好的，谢谢助教

closed 2025-01-02 by zZzZ9zZ9