pku-liang/hlcd-spmm-project #23: spmm中rhs_data显示还是有问题

wwwqwq2022, 2024-12-24:
我尝试了
[image]
还是mei没有display,感觉是用了二维压缩数组的原因导致这个显示不出来还是自己的程序有问题
[image]

wwwqwq2022, 2024-12-25:
顺便问一下这个60分的部分是只需要生成16个的加法器和乘法器，还是要生成N个加法器和乘法器？

KEKE046, 2024-12-25:
这个好像确实不支持，是 verilator 的问题。可以用 `$display` 手动打出来显示。

KEKE046, 2024-12-25:
可以在代码里放四个 `logic`，这样可能 `gtkwave` 会比较开心
```systemverilog
output logic [7:0] rhs_data_0[`N-1:0];
output logic [7:0] rhs_data_1[`N-1:0];
output logic [7:0] rhs_data_2[`N-1:0];
output logic [7:0] rhs_data_3[`N-1:0];
assign rhs_data_0 = rhs_data[0];
assign rhs_data_1 = rhs_data[1];
assign rhs_data_2 = rhs_data[2];
assign rhs_data_3 = rhs_data[3];
```

closed 2024-12-25 by wwwqwq2022