pku-liang/hlcd-spmm-project #53: 关于dbbuf仿真测试的疑问

angerybob, 2025-01-15:
[image]
[image]

请问为什么有的测试样例out_start变成高电平后一周期仿真就立刻结束了，按道理start之后还应该持续N/4个周期吧。这时候结束，输出都还没接收完吧？
输出结果如下图所示，会导致最后四行还没接收到输出就测试就结束了
[image]
只有个别样例会有这个现象，其余可以正常输出

KEKE046, 2025-01-15:
看你的波形图，确实是输出了4个周期，verilator 最后一个周期会在上升沿瞬间结束，所以没有显示
[image]

angerybob, 2025-01-16:
[image]
但是我看大部分正常的波形图都会在第四周期末尾才结束？

angerybob, 2025-01-16:
虽然不知道什么原因，但是现在好像不会了

closed 2025-01-16 by angerybob